iobroker-community-adapters/ioBroker.beckhoff #3: TC2 Global_Variables are not recognized

NilsTC, 2018-11-05:
TC2 needs a "." at the beginning.

dkleber89, 2018-11-11:
Are u firm with node, npm and Beckhoff?

Is it possible for u, that u use this Project: https://github.com/dkleber89/beckhoffTest on a Testsystem in combination with a TwinCat2 PLC and give me back the Console Log Output?

I need to know how variables exactly looks like.

Im not able to run a TwinCat2 Runtime on a x64 System and i have no PLC or x86 System here.

NilsTC, 2018-11-18:
Not really....
I just tried it in VS to start.
It also comes out a debugg file.
I can not find useful information in it. Probably I'm doing something wrong.

How should I proceed?

dkleber89, 2018-11-20:
I have the needed Information yet ... I see that i need to change many things for TC2

commit 5acb3e29 (2018-11-21): Merge pull request #4 from dkleber89/bugfix-3

bugfix-3 fix #3
closed 2018-11-21 by dkleber89