iobroker-community-adapters/ioBroker.beckhoff #10: Think about to fix the issues found by adapter checker

ioBrokerChecker, 2019-03-12:
I am an automatic service that looks for possible errors in ioBroker and creates an issue for it. The link below leads directly to the test:

https://adapter-check.iobroker.in/?q=https://raw.githubusercontent.com/dkleber89/ioBroker.beckhoff

- [ ] [E021] "licenses" in package.json are depricated. Please use only "license": "NAME" field.
- [ ] [E140] width and height of logo are not equal
- [ ] [E301] Tests on Travis-ci.org are broken. Please fix.
- [ ] [E605] No actual year found in copyright. Please add "Copyright (c) 2019 [object Object]" or "Copyright (c) 20xx-2019 [object Object]" at the end of Readme
- [ ] [E701] No actual year found in LICENSE. Please add "Copyright (c) 2019 [object Object]" or "Copyright (c) 20xx-2019 [object Object]" at start of LICENSE

I have also found a few warnings that may be fixed, if possible.

- [ ] [W113] Adapter should support compact mode

Thanks,
your automatic adapter checker.

P.S.: There is a community in Github, which supports the maintenance and further development of adapters. There you will find many experienced developers who are always ready to assist anyone. New developers are always welcome there. For more informations visit: https://github.com/iobroker-community-adapters/info

dkleber89, 2019-03-12:
All Points are solved ... W113 duplicates Issue #8

closed 2019-03-12 by dkleber89

ioBrokerChecker, 2019-03-28:
Thanks, that all bugs have been fixed.