iobroker-community-adapters/ioBroker.beckhoff #12: The driver does not detect a hard reset (power loss) plc.

Andrey2509, 2019-03-29:
The driver does not detect a hard reset (power loss) plc (CX9010). 
"info.сonnection" and "info.plcRun"  stay with status "true".
PLC objects are not synchronized. To solve the problem, you have to restart the driver.

dkleber89, 2019-03-29:
Is this reproducible? How long was the power loss?

Andrey2509, 2019-03-30:
I repeated the situation using the "Aсtivate Configuration..." from System Manager. Objects stop updating their values.

dkleber89, 2019-03-30:
Okay another 2 Questions:

Did u change some Values in ioBroker at this time e.g. in a Skript or not?
When u wait maybe 5 min then the Adapter reconnect or stay in this undefined State?

Andrey2509, 2019-03-30:
No values of objects in iobroker did not change.
After restarting the PLC (via  "Aсtivate Configuration..." from System Manager), the driver does not update values of objects, waiting for more than 30 minutes does not help and "info.сonnection" and "info.plcRun" stay with status "true". Only manually restart driver helps.

dkleber89, 2019-03-30:
Ok thanks...I have an Idea what the Problem is, i will investigate it ...In the meantime maybe when u have time you can try something?...Try to change in the undefined State one object in ioBroker and look whats happen...

Andrey2509, 2019-03-30:
1 Before 15:15:35 Data from plc sync ok: "Object: .IOBROKER.CHILD_BIG_ROOM_CO2 -> 616"
2. At 15:15:35 I restart the PLC (via "Aсtivate Configuration..." from TC2 System Manager)
3. At 15:16:13.613 plc fully restarted and you can see mqtt client on plc attached to iobroker mqtt server again.
4. At 15:16:59.942 I changed  .IOBROKER.TARGET_CO2 -> 400, This value successfully get PLC. 
5. No new data from plc after 15:15:35.

Debug log :

`

beckhoff.0 | 2019-03-30 15:27:42.703 | debug | State of PLC: RUN
-- | -- | -- | --
beckhoff.0 | 2019-03-30 15:27:27.700 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:27:12.699 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:26:57.698 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:26:42.698 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:26:27.690 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:26:12.689 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:25:57.681 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:25:42.679 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:25:27.675 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:25:12.672 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:24:57.668 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:24:42.667 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:24:27.668 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:24:12.665 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:23:57.649 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:23:42.647 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:23:27.642 | debug | State of PLC: RUN
web.0 | 2019-03-30 15:23:22.589 | info | ==>Connected system.user.admin from ::ffff:192.168.3.128
web.0 | 2019-03-30 15:23:22.262 | info | <==Disconnect system.user.admin from ::ffff:192.168.3.128 flot
beckhoff.0 | 2019-03-30 15:23:12.643 | debug | State of PLC: RUN
web.0 | 2019-03-30 15:23:10.220 | info | ==>Connected system.user.admin from ::ffff:192.168.3.148
web.0 | 2019-03-30 15:23:10.020 | info | <==Disconnect system.user.admin from ::ffff:192.168.3.148 flot
beckhoff.0 | 2019-03-30 15:22:57.631 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:22:42.627 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:22:27.623 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:22:12.624 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:21:57.623 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:21:42.621 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:21:27.605 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:21:12.603 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:20:57.601 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:20:42.601 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:20:27.589 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:20:12.576 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:19:57.574 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:19:42.568 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:19:27.843 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:19:12.564 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:18:57.546 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:18:46.993 | debug | Write Symbol Value from ioBroker to PLC: .IOBROKER.TARGET_CO2 -> 500
beckhoff.0 | 2019-03-30 15:18:42.541 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:18:27.533 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:18:12.533 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:17:57.521 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:17:42.506 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:17:27.503 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:17:12.503 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:16:59.942 | debug | Write Symbol Value from ioBroker to PLC: .IOBROKER.TARGET_CO2 -> 400
beckhoff.0 | 2019-03-30 15:16:57.502 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:16:42.502 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:16:27.500 | debug | State of PLC: RUN
mqtt.0 | 2019-03-30 15:16:13.613 | info | Client [plc] reconnected. Old secret 1553916920909_1643. New secret 1553940973217_8024
beckhoff.0 | 2019-03-30 15:16:12.502 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:15:57.500 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:15:42.500 | debug | State of PLC: RUN
beckhoff.0 | 2019-03-30 15:15:35.084 | debug | Write Symbol Value from PLC to ioBroker: .IOBROKER.CHILD_BIG_ROOM_CO2 -> 616
beckhoff.0 | 2019-03-30 15:15:34.938 | debug | Write Symbol Value from PLC to ioBroker: .IOBROKER.KITCHEN_ROOM_CO2 -> 639
beckhoff.0 | 2019-03-30 15:15:33.235 | debug | Write Symbol Value from PLC to ioBroker: .IOBROKER.KITCHEN_ROOM_HUM -> 30.799999237060547
beckhoff.0 | 2019-03-30 15:15:27.500 | debug | State of PLC: RUN

`

>

dkleber89, 2019-04-01:
I try to reproduce ... It seems to be a Problem with TC2 Systems ... On my TC3 System im not able to reproduce it ... unfortunately i don´t have a TC2 System for Testing so it´s heavy for me to debug that and find a proper Solution.

Please be patient i need a little Time to find a Solution for that

dkleber89, 2019-04-01:
Another Idea ... can u try this? -> 

Can u try to write a Tag from ioBroker to PLC between the restart and the reconnect of MQTT and tell me what happen then?

Thanks for ur help

dkleber89, 2019-04-18:
Added a Monitoring to the Ack State ... When you change a Variable and then no Ack coming from PLC the Adapter trigger a resync and everything is running again ... It only Works when u change Something on ioBroker otherwise not ... You can make a Wachdog Variable in PLC and change it in ioBroker in some Interval the it is correctly monitored ... I will add that in the Future builtin to the Adapter

closed 2019-04-18 by dkleber89