iobroker-community-adapters/ioBroker.beckhoff #15: A large number of values are not transferred from PLC to iobroker

Andrey2509, 2019-07-15:
1. A large number of values are not transferred from PLC to iobroker.
No problem if the number of values are 64 (driver in "debug" level) - I did not detect losses during transmission.
But if you pass 82 values, then some of the values are not transferred (nothing in iobroker debug logs for updated values from PLC).

2. I found that in the "debug" level of the driver in iobroker, the transfer of values is stable (64 values). In the "info" level (64 values), not all values are transferred from PLC to the ioBroker.
P.S. I use TC2, beckhoff CX9000.

dkleber89, 2019-07-23:
1. When i take a look on the Code from the Adapter i see no Reason why i should have these Borders on this Side. Maybe it´s a Boarder of the Twincat 2 System? 64 is a special Number it´s not really Bug like -> Maybe there are some Settings in Twincat 2 System?

One thing i can see is that when on the NOTIFY Registration an Error occure -> This error not write to log. I change this behavior this evening and push a new Version to NPM. 

2. It´s per Design impossible that the Log-Level has any impact to the functionality of the Adapter. When u have seen this behavior then it was only coincidence.

dkleber89, 2019-07-24:
You should be able to install the Version 1.0.3 over the latest Repo in ioBroker. Hopefully you can see yet maybe some Problem when u run the Adapter in Debug Mode. I make Today a Test with 80 Variables and i have no Problem. It was on TwinCat 3 but for the Adapter and the ADS Protocol there is no difference. Maybe there is a Problem with TC2 (Functionality or Settings) but i'm sorry i can't help you with that, because i have not much experince with TC2.

Andrey2509, 2019-07-30:
Now (ver 1.0.3) I got  logs with random values. All values have data on PLC (BOOL or REAL):

beckhoff.0 | 2019-07-30 19:58:08.445 | warn | Subscription for: .IOBROKER.M10T1 occure an Error: undefined
beckhoff.0 | 2019-07-30 19:58:08.435 | warn | Subscription for: .IOBROKER.M9T2 occure an Error: undefined
beckhoff.0 | 2019-07-30 19:58:08.435 | warn | Subscription for: .IOBROKER.M9T1 occure an Error: undefined
beckhoff.0 | 2019-07-30 19:58:08.435 | warn | Subscription for: .IOBROKER.M8T2 occure an Error: undefined

I don't understand reason for  "Error undefined"  (and these values  not synced from PLC)...

dkleber89, 2019-08-01:
Unfortunately i have in Version 1.0.3 a Typo. But on the other Side on a short Test the Error Message is yet also not really better '... Error: Error'.

I have a Question when u try it again always the same Variables have the Error or it's random?

I don't understand what the Problem is or what the Problem can be ... On my actual Standpoint i can only say that i don't think that is a Problem from the Adapter -> A Number of Variables working fine with your PLC -> And i can not see anything on Adapter that could have a Problem with the amount of Variables especially when it works with big amounts on other PLC's.

Maybe you should ask the Beckhoff Support? Maybe there is some Reason that not more Variables can be subscribed.

I'm sorry that i can't help you in these Case ...

dkleber89, 2019-08-13:
Closed ... Nothing to do at this Time ... Feel free to reopen it when u have new Things to discuss

closed 2019-08-13 by dkleber89