iobroker-community-adapters/ioBroker.beckhoff #33: Is it currently possible to synchronize more than one variable

PLCHome, 2020-06-19:
I am trying to replace my Node Red with ioBroker. Hence the question, is it currently possible to synchronize more than one variable between iobroker and Twincat?

I only found the input for a variable.

PLCHome, 2020-07-07:
?

dkleber89, 2020-07-15:
Oh sorry for the late response ... i don't get a notification from Github ... Yes for sure its possible ... The input is not for a Variable ... in TwinCat 3 it's for a Global Variable Table and in TwinCat 2 its for a global Struct

dkleber89, 2020-07-15:
The Adapter syncronises all Variables automatic in the ioBroker (Complete Tree) but only when all Variables inside are supported ... See readme

PLCHome, 2020-07-15:
So only one variable that contains a structure can be registered.
.VarStruct1
.VarStruct2
it does not work.

dkleber89, 2020-07-17:
Yes that is true

PLCHome, 2020-07-17:
What do you think if I convert the configuration to tab and install a tab in which you can specify more than one variable table?
The currently maintained variable could be transferred to the first line. I would implement it and make a merge request.

It could look like this:

[image]

dkleber89, 2020-07-17:
That's a good idea ... It's would be a good feature and a good improvement for the Adapter to use more than one Struct/V-Table

dkleber89, 2020-07-17:
I would be happy when u implement this ... Looking forward for your merge request

Timmes123, 2020-10-27:
Hi :-) i work with your adapter and for me it would be great, if the adapter get this update.
I have a lot variables i have to read from existing PLCs, and i dont want to change software in every PLC and integrate a interface vartable. 
it would be great, if i can define the needed variables in iobroker, without touching the plc

dkleber89, 2021-06-13:
@PLCHome ... Are u working on that? Or is this issue obsolete?

dkleber89, 2023-03-21:
Close because stale

closed 2023-03-21 by dkleber89