iobroker-community-adapters/ioBroker.beckhoff #34: Verliert Verbindung

zonfacter, 2020-10-29:
Hallo,
Folgendes Problem seit ich in dem ioBroker alle Adapter aktualisiert habe. (25.10.2020)

Kommt es verhäuft am Tag vor das die Verbindung zur Beckhoff Steuerung über deinem Adapter nicht mehr stattfindet.
Es werden keine Datenpunkte aktualisiert im Iobroker(VM)<->Beckhoff (CX8090 TC2) (Keine W-Lan Verbindung)

2020-10-28 06:11:09.688 - warn: beckhoff.0 (9507) Subscription for: .iobroker.Rolladen.RollerBlind_5 occure an Error: timeout

Wenn die Verbindung hergestellt ist, funktioniert auch alles sofort. Eingabe -> Ausgabe ist ~ Zeitgleich.

Hast du eine Idee woran das liegen, bzw. wo ich evtl. den Fehler finden kann ?

Sollte ich sonst ein Watchdog machen ? Bzw. in meine Skripte mit integrieren ?

Info:
Platform: linux
os: linux
Architecture: x64
CPUs: 2
Speed: 2195 MHz
Model: Intel(R) Xeon(R) CPU E31220L @ 2.20GHz
RAM: 2.9 GB
System uptime: 3 daysShortText 17:46:03
Node.js: v12.13.1
NPM: 6.12.1
adapters count: 371
Disk size: 60.3 GiB
Disk free: 46.8 GiB
Uptime: 3 daysShortText 12:03:05
Active instances: 39

dkleber89, 2020-10-29:
Ein Watchdog kann sicher helfen herauszufinden wann diese unterbrechnungen auftreten um ein Muster zu erkennen

zonfacter, 2020-10-30:
So ich habe erstmal bei meiner VM den Snapshot wiederhergestellt wo alles Funktioniert hat.
(Was bin ich froh das ich einen Snapshot gemacht hatte :-D )

Einen Watchdog habe ich trotzdem eingerichtet.

Lasse mir dann eine Telegramm Nachricht zu kommen und schreibe es in die iobroker LOG.
Kann mir das nicht erklären.

Ob es an dem Update von den Adaptern liegt:
Admin 4.1.6 -> 4.1.10
socket.io 3.0.12 -> 3.0.13

Hier die Auflistung der Adapter die alle upgedatet wurden:
  Admin  4.1.6    4.1.10   
  BackItUp  1.6.9    1.7.0   
  Das Wetter.com  3.0.1    3.0.4   
  deConz ZigBee  1.3.11    2.0.5   
  DS18B20  1.1.4    1.1.5   
  HABpanel  0.4.1    0.4.3   
  HomeMatic ReGaHSS  2.6.22    2.6.23   
  HomeMatic RPC  1.14.23    1.14.24   
  Philips Hue Bridge  3.3.5    3.3.8   
  iQontrol  1.1.11    1.3.6   
  Script Engine  4.6.23    4.9.7   
  Visualization with Lovelace-UI  1.1.0    1.2.7   
  Material UI  0.13.8    0.13.9   
  minuVis  0.0.6    0.0.10   
  Mobile UI  0.5.0    1.0.1   
  Screenshots with PhantomJS  1.1.0    1.1.2   
  PING  1.4.11    1.4.12   
  shuttercontrol  1.0.2    1.0.4   
  Simple RESTful API  2.4.7    2.4.8   
  Smart Control  0.5.6    0.5.12   
  socket.io  3.0.12    3.0.13   
  SourceAnalytix  0.4.7-beta.0    0.4.8-alpha.2   
  SQL History  1.15.2    1.15.3   
  Visualisation  0.1.1    0.3.1   
  Visualisation  1.2.4    1.3.4   
  hqwidgets style Widgets  1.1.5    1.1.6   
  inventwo Design Widgets  2.2.0    2.5.3   
  WEB server  3.0.11    3.0.13   
  Zigbee  1.2.1    1.3.0

zonfacter, 2020-10-30:
hmmm

beckhoff.0 | 2020-10-30 11:41:08.552 | error | (15490) ADS Client: Error [ERR_STREAM_WRITE_AFTER_END]: write after end
-- | -- | -- | --
beckhoff.0 | 2020-10-30 11:41:08.548 | error | (15490) ADS Client: Error [ERR_STREAM_DESTROYED]: Cannot call write after a stream was destroyed
beckhoff.0 | 2020-10-30 11:41:08.077 | info | (15490) Try to reconnect in 15 seconds
beckhoff.0 | 2020-10-30 11:41:08.076 | error | (15490) ADS Client: Error: This socket has been ended by the other party
beckhoff.0 | 2020-10-30 11:41:06.763 | info | (15490) Connected to PLC. State of PLC: RUN
beckhoff.0 | 2020-10-30 11:40:51.745 | warn | (15490) Write Symbol Value from ioBroker to PLC failed: timeout
beckhoff.0 | 2020-10-30 11:40:51.745 | warn | (15490) Write Symbol Value from ioBroker to PLC failed: timeout
beckhoff.0 | 2020-10-30 11:40:51.744 | warn | (15490) Write Symbol Value from ioBroker to PLC failed: timeout
beckhoff.0 | 2020-10-30 11:40:51.744 | warn | (15490) Write Symbol Value from ioBroker to PLC failed: timeout
beckhoff.0 | 2020-10-30 11:40:51.743 | warn | (15490) Write Symbol Value from ioBroker to PLC failed: timeout
beckhoff.0 | 2020-10-30 11:40:51.739 | warn | (15490) Write Symbol Value from ioBroker to PLC failed: timeout
beckhoff.0 | 2020-10-30 11:40:51.738 | warn | (15490) Write Symbol Value from ioBroker to PLC failed: timeout
beckhoff.0 | 2020-10-30 11:40:51.738 | warn | (15490) Write Symbol Value from ioBroker to PLC failed: timeout
beckhoff.0 | 2020-10-30 11:40:51.738 | warn | (15490) Write Symbol Value from ioBroker to PLC failed: timeout
beckhoff.0 | 2020-10-30 11:40:51.737 | warn | (15490) Write Symbol Value from ioBroker to PLC failed: timeout
beckhoff.0 | 2020-10-30 11:40:51.737 | info | (15490) Try to reconnect in 15 seconds
beckhoff.0 | 2020-10-30 11:40:51.735 | warn | (15490) Write Symbol Value from ioBroker to PLC failed: timeout

zonfacter, 2020-10-30:
So ich habe jetzt mal testweise meine kontinuierliche Kommunikation per ADS eingestellt. (aktueller Stromverbrauch)
Nur noch reine Taster Impulse..

Jetzt zur Arbeit fahren.

zonfacter, 2020-11-01:
So seit dem die Dauer Kommunikation deaktiviert wurde iobroker<->CX8090 Beckhoff gibt es auch keine Timeouts mehr.
Eigentlich sollte es doch dabei keinerlei Probleme geben. Es war sogar stellenweise soweit gekommen das der CX keinerlei Login (PLC) mehr ermöglicht hat. Auch der MODBUS Service auf CX8090 hat sich dann aufgehängt. Webservice zum Neustarten des CX funktionierte wieder rum.

Gibt es es auf deiner Seite noch Möglichkeiten eine bessere Diagnose zu fahren ? (Beckhoff System Manager hat nix angezeigt - da keine Kommunikation mehr möglich war ADS)

dkleber89, 2020-11-01:
Hast du sonst noch was verändert? ... Dadurch das es jetzt auf einmal mit dem alten Snapshot auch nicht mehr geht würde ich mal vorsichtig sagen es hat nix mit dem ioBroker und dem Adapter zu tun ... Was meinst du mit Dauerkommunikation?

zonfacter, 2020-11-02:
Hallo,
Mit Dauerkommunikation ist gemeint. 
Das mein Stromverbrauch / Wasserverbrauch / Gasverbrauch der in der PLC errechnet wird, bei jeder Veränderung per ADS Verbindung zum iobroker übertragen wird. (Der Baustein in der PLC hat eine UPDATE_TIME := t#1s;) Gibt es bei deinem Adapter irgendwelche Limitationen welche dies beeinflussen könnte ?
 

_CX8090 Werte
Die PLC Tasktime läuft auf 10ms. (tatsächliche Zyklus Geschwindigkeit 2.7ms)
Basiszeit beläuft sich auf 1ms
Echtzeitauslastung 10%_

Wie gesagt seitdem ich die jetzt aus kommentiert habe, gibt es keine LOG Einträge mehr.

(*
iobroker.Verbrauch :=		 	Wasserflow.Stromverbrauch.F	;	(* Momentan Verbrauch Strom *)
iobroker.Impuls_Strom:=		bTaster31			;			(* Impulsgeber Strom *)

iobroker.Verbrauch_Wasser :=	 MAIN.Volume	;	(* Momentan Verbrauch Wasser *)
iobroker.Impuls_Wasser:=		bTaster30			;			(* Impulsgeber Wasser *)

iobroker.Verbrauch_Gas:=		MAIN.GAS_dwCounterValue + MAIN.GAS_CounterPostComma	;			(*  Momentan Verbrauch Gas *)
iobroker.Impuls_Gas:=			bTaster2				;			(* Impulsgeber Gas *)
*)

In der Beckhoff Steuerung habe ich die letzte änderung im Mai gemacht.
Rolladen Steuern (0-100%) Vorgabe - Visu über iobroker ads 
PLC Rückgabe = (Richtungsvorgabe / Stop) -> iobroker ADS -> RFLINK -> Rolladen Empfänger
Da gibt es keinerlei Probleme.

OffTopic:
Ich habe interessanterweise auf Arbeit an einer Montage-Linie - ADS Verbindungsabbrüche gehabt. Da lag es am internen Maschinen-Netzwerkswitch.  Verfolgt mich gerade 🤣

dkleber89, 2020-11-03:
Zu den Limits ... Solange die Maschine auf der der ioBroker läuft genug Leistung hat gibts von daher keine Limits. Der Adapter ist natürlich nur single Threaded also mehr Kerne bringen nix. Aber von solchen Grenzen sind wir hier ganz sicher weit weg.

Wenn du testweise diese Daten alle 2 oder 5 Sekunden aktualisiert bleibt dann das Problem bestehen?

zonfacter, 2020-11-03:
Ich werde es Heute Abend / Morgen mal umsetzen. (Alle 5 Sekunden reicht zur Visualisierung 😄  )

Leistungstechnisch: Intel(R) Xeon(R) CPU E31220L @ 2.20GHz reicht eigentlich für alles was man in der Hausautomation sich vorstellen kann.

zonfacter, 2020-11-03:
So hab es eingepflegt alle 5 Sekunden für 10ms (1 Zyklus) werden die werte Aktualisiert.

zonfacter, 2020-11-05:
So ich habe keine LOG Einträge mehr.
Ich denke wir können hier schließen.
Falls irgendwer mal die gleichen / ähnlichen Probleme hat kann man dies ja hier nachverfolgen.

Danke für deine Kommunikative Unterstützung :-D , manchmal braucht man nur jemanden zum reden um auf die Lösung zu kommen.

zonfacter, 2020-11-08:
Moin dkleber89,

Ich war wieder fleißig.
Und erhalte wieder Timeout Meldungen.
Was habe ich gemacht.
Neue Strukturen angelegt in der PLC
Mit jede Menge neue Variablen ~ 160 Neue Variablen. In Summe 219.

Jetzt habe schrittweise wieder alles aus kommentiert. Und keine Timeout Meldung.
Gibt es eine Limitierung an Variablen die du abfragen kannst ?
Ich habe zurzeit 59 Funktionierte Variablen.  
....

So hab mal getestet sobald ich mehr wie >60 Variablen habe kommt es zu einem Timeout.
Nach starten des Adapter.

dkleber89, 2020-11-10:
Hallo ... Ich kann dir nicht wirklich sagen was bei dir los ist ... Ich kann nur sagen weder die "schnelle" Aktualisierung noch die Variablenanzahl wird hier ein Problem darstellen. Hatte bisher noch nicht eine Rückmeldung die irgendwie auf eine schlechte Performance hindeutet. Ich glaube du musst hier an einem anderen Punkt suchen...

closed 2020-11-10 by dkleber89