iobroker-community-adapters/ioBroker.beckhoff #37: ADS Mailbox overflow Start / Stop Adapter

NilsTC, 2021-01-04:
Hi,

vor allem bei Starten und Stoppen des Adapters aber auch manchmal zwischendurch werden sehr viele Anfragen an die SPS versendendet. 
Das hat zur Folge, dass die ADS Mailbox überläuft bzw. voll ist und das wirkt sich leider auch auf andere Prozesse / Kommunikationen der SPS aus. z.B. angeschlossene Buskoppler BK9100 oder auch z.B. die Modbus TCP Kommunikation.
Diese geht dann unter umständen kurzzeitig in einen Fehler.

Ich habe zu dem Thema ein Dokument und einen Bereich im Infosys auf der Beckhoff Website gefunden welches das Thema ein bisschen erklärt:
https://download.beckhoff.com/download/document/application_notes/dk9322-0413-0070.pdf
https://infosys.beckhoff.com/content/1031/tcadscommon/html/tcadscommon_cookbookadsclient.htm

Kannst du dir das Thema mal anschauen ob sich das noch Optimieren lässt?
Vielleicht kann etwas Zeitlich verzögert werden oder mit den bei Beckhoff beschriebenen "ADS Sum Command" gearbeitet werden.
Gerne kann ich auch behilflich sein, wenn du einen Ansatz hast!?

Der Vollständigkeit hier noch ein Paar Infos zu den Versionen auf meinem Testsystem.
Spielt aber glaube ich keine Rolle.
 - Adapter Version: 1.2.2
 - Beckhoff Hardware CX8190 TC3 4024.7
 - Beckhoff XAE TC3 4024.11

dkleber89, 2021-06-13:
Hallo @NilsTC 

ich denke das Problem passiert während der Adapter sich auf die Variablenänderungen der SPS registriert. Ich habe derzeit nicht wirklich Zeit mich besser darum zu kümmern ... Ich habe eine Änderung gemacht https://github.com/dkleber89/ioBroker.beckhoff/tree/fix/subscribe-valuechange-plc ... Wenn du Zeit hast wäre es super wenn du das Testen könntest (Über die Installation von GitHub im ioBroker)... Ich habe nur schnell ins blaue programmiert und noch garnix getestet

Eine Sache ist auch noch -> Ich weiß nicht genau ob es mit dieser Lösung nicht ins andere extreme schwenkt und sehr langsam geht.

NilsTC, 2021-09-03:
Hi,
wir haben die Änderung an drei verschiedenen Projekten mit TC3 und TC2 getestet und bei allen hatte wir jetzt keine Fehler mehr in der Start Phase.
Diese Dauert zwar ein bisschen länger danach läuft aber alles Flüssig und schnell.

Also könnte die Anpassung gerne in die Hauptversion mit eingebaut werden.

dkleber89, 2021-09-28:
Done #51

closed 2021-09-28 by dkleber89