iobroker-community-adapters/ioBroker.beckhoff #54: Sync PLC Variables: Error: Type (TIME) found but not Supported!!

dema0fzj, 2021-08-20:
Moin,

mein Beckhoff ADS Adapter meldet seit einiger Zeit folgenden Fehlermeldung und Synchonisiert die GVL nicht mehr:

Sync PLC Variables: Error: Type (TIME) found but not Supported!! Please Check your Variable Table and your Datatypes in PLC

In der GVL habe ich allerdings keine Variable des Type (TIME) drin.

Folgende Schritte habe ich bislang Probiert:
- GVL auf Datentypen geprüft -> kein Erfolg
- Neue GVL mit anderem Namen und zwei Bool Variablen erstellt und im Adapter hinterlegt. -> kein Erfolg
- Update des Beckhoff ADS Adapters auf V1.4.1 -> kein Erfolg
- Löschen der Objekte in ioBroker und anschließender Neustart der Beckhoff Instanz -> kein Erfolg
- Deintsallation des Beckhoff Aapters, anschließene Neuinstallation -> kein Erfolg
- Bereinigen des PLC Projektes, Download des PLC Projektes mit vollständigem Laden  -> kein Erfolg

Ich habe die Kommunikation zwischen ioBroker und PLC redundant aufgebaut, die Kommunikation über eine zweite, "eigentlich Identische" ioBroker installation macht keine Probleme mit dem Beckhoff Adapter.

Hat jemand schon mal ähnliche Probleme gehabt? Würde mich über Ideen freuen um die Herkunft des Problems zu finden. 

Gruß, Dennis

GVL:
[image]

Adaptereinstellungen:
[image]

dkleber89, 2021-09-28:
ist das noch aktuell?

dema0fzj, 2021-09-28:
Ja, das Problem ist noch vorhanden.

dkleber89, 2021-10-07:
Es laufen zwei ioBroker Instanzen und eine davon läuft und die andere nicht? 

Kann mir keinen reim darauf machen was da das Problem sein soll ... Ich würde (nochmal) folgendes Versuchen:
- Alle Beckhoff Adapter Instanzen löschen
- Die Beckhoff Adapter Installation auch löschen (Im Bereich Adapter -> Adapter löschen
- Alles was in irgendeiner Form vom Beckhoff Adapter kommt aus den Objekten löschen (Expertenmodus aktivieren um alles zu sehen)
- In der ioBroker Ordnerstruktur noch nachschauen ob es irgendwelche Beckhoff Adapter leichen gibt
- ioBroker neu starten
- Adapter regulär installieren und konfigurieren

Frage die zwei ioBroker Instanzen laufen nicht auf dem selben host oder?

dkleber89, 2021-11-04:
Close -> Stale -> Feel free to reopen when needed

closed 2021-11-04 by dkleber89