iobroker-community-adapters/ioBroker.beckhoff #98: adsClientGetSymbolsError: timeout

pain33322, 2023-04-20:
**Describe the bug**  
Hallo zusammen,

ich habe ein Problem mit dem Ads Adapter für Beckhoff in IOBroker. Ich habe bereits zwei Instanzen von IOBroker im Einsatz und habe keine Probleme damit. Allerdings bekomme ich bei der dritten Instanz immer wieder folgende Fehlermeldung:

[image]

Ich habe bereits die Konfiguration meines IOBroker-Systems sowie die Route zur Verbindung mit dem Beckhoff-Steuerungssystem überprüft und konnte keine Fehler feststellen. Allerdings muss ich erwähnen, dass ich für diese Verbindung einen nicht ganz so schnellen VPN nutze. Aufgrund dessen vermute ich, dass das Timeout-Problem mit der VPN-Verbindung zusammenhängen könnte, bin mir jedoch nicht sicher.

Aus dem Quellcode vom "node-ads-api" konnte ich entnehmen, dass es die Timeout-Variable "timeout" gibt, welche nicht im adapter überschrieben wird.

```
var options = {
    //The IP or hostname of the target machine
    host: "10.0.0.2",
    //The NetId of the target machine
    amsNetIdTarget: "5.1.204.160.1.1",
    //The NetId of the source machine.
    //You can choose anything in the form of x.x.x.x.x.x,
    //but on the target machine this must be added as a route.
    amsNetIdSource: "192.168.137.50.1.1",

    //OPTIONAL: (These are set by default)
    //The tcp destination port
    //port: 48898
    //The ams source port
    //amsPortSource: 32905
    //The ams target port for TwinCat 2 Runtime 1 
    //amsPortTarget: 801 
    //The ams target port for TwinCat 3 Runtime 1
    //amsPortTarget: 851 
    //The timeout for PLC requests
    //timeout: 500
    //The Local address the socket should connect from
    //localAddress: "192.168.137.50"
    //The Local port the socket should connect from
    //localPort: 50000
    //Version of IP stack. Must be 4, 6, or 0. The value 0 indicates that both IPv4 and IPv6 addresses are allowed. Default: 0
    //family: 4
}
```

Ich vermute, dass mein Problem damit zusammenhängen könnte. Deshalb würde ich gerne wissen, ob es möglich ist, die Timeout-Variable in der "node-ads-api" von 500 ms auf 10000 ms zu erhöhen, um zu testen, ob dies das Problem lösen kann. 

**Versions:**  
 - Adapter version: 1.5.2
 - Node version: 5.1.25
 - Operating system: linux

dkleber89, 2023-04-27:
Hallo, diese Variable wird im Adapter nicht beschrieben und kann somit aktuell auch nicht einfach zum Testen erhöht werden. Das müsste man implementieren

pain33322, 2023-04-28:
> Hallo, diese Variable wird im Adapter nicht beschrieben und kann somit aktuell auch nicht einfach zum Testen erhöht werden. Das müsste man implementieren.
> 
> Aber je nach Situation könnte es theoretisch dein Problem lösen wobei ich 500ms jetzt nicht so wenig finde. Aber kommt auf die Verbindung und natürlich auf die Größe des Programmes in der PLC an.

Hallo zusammen, 

ich habe jetzt die Änderung der Timeout-Variable mit einer Kopie des Adapters in meinem iobroker umgesetzt. Es hat sofort funktioniert.

[image]

Zur Info ich habe einen Ping zwischen den beiden Standorten von c.a 60ms allerdings ist die Bandbreite mit DSL16(11Down/3Up) nicht so das wahre. Eine Anpassung in meinem Fall auf 2000 hat den Fehler bereits behoben. Ich wäre sehr dankbar, wenn in zukünftigen Versionen die Funktion zur Anpassung dieser Variable in der Konfigurationsoberfläche  implementiert werden könnte. 

Liebe Grüße

skiworker, 2024-10-09:
Hallo Zusammen,
@pain33322 
Kannst Du mir sagen, wie Du das gemacht hast? 
Ich habe ein ähnliches Problem und vermute, dass die Erhöhung des Timeouts auch mein Problem lösen wird.

Liebe Grüße

zonfacter, 2026-10-13:
Die Einstellung ist in [1.8.0-beta.2 fork prerelease](https://github.com/zonfacter/ioBroker.beckhoff/releases/tag/v1.8.0-beta.2) umgesetzt ([implementation PR](https://github.com/zonfacter/ioBroker.beckhoff/pull/1)): „ADS request timeout (ms)“ / `adsTimeout`, Standard 500 ms, einstellbar von 100 bis 60000 ms. Für den beschriebenen VPN-Fall kann zunächst 2000 ms getestet werden; auch 10000 ms ist möglich. Der Wert wird an node-ads-api übergeben und ist vom Reconnect-Intervall getrennt.

Zusätzlich werden bei kurzen Unterbrechungen validierte Symbolmetadaten wiederverwendet; ADS-Handles/Notifications werden neu registriert und aktuelle Werte gelesen. Simulator-Tests prüfen Timeout/Reconnect. Die Beta läuft auf einer bestehenden TwinCAT-2-Installation; ein VPN-Test beim ursprünglichen Melder steht noch aus. Dies ist ein Fix im Fork, noch keine Veröffentlichung im Originaladapter.

Proposed upstream in [PR #445](https://github.com/iobroker-community-adapters/ioBroker.beckhoff/pull/445). [Successful fork CI](https://github.com/zonfacter/ioBroker.beckhoff/actions/runs/38462698504). Upstream integration/closure remains with the maintainers.

mcm1957, 2026-10-19:
should be solved with 2.0.0.

If probelem still exists please open a new issue and attach a debug log

closed 2026-10-19 by mcm1957