iobroker-community-adapters/ioBroker.beckhoff #224: Allow Hostname, not only IP Adress

Timmes123, 2024-10-09:
**Describe the bug**  
The Adapter just allow to save when there is a IP Adress in the Target Device Adress Field
When i try to use the Hostname like "Devicename" or "Devicename.domain.local" its not possible to press the save Button

**To Reproduce**  
Type in something different

**Expected behavior**  
Allow also Hostname.

**Screenshots & Logfiles**  
Not necessary

**Versions:**  
 - Adapter version: <1.6.0>
 - JS-Controller version: <6.0.11> 
 - Node version: <v18.17.1> 
 - Operating system: <ubuntu>

**Additional context**  
Not necessary

zonfacter, 2026-10-13:
Hostname support is implemented in [1.8.0-beta.2 fork prerelease](https://github.com/zonfacter/ioBroker.beckhoff/releases/tag/v1.8.0-beta.2) ([implementation PR](https://github.com/zonfacter/ioBroker.beckhoff/pull/1)). The target-address field accepts IPv4 addresses and hostnames such as `Devicename` and `Devicename.domain.local`; the hostname is passed through to the ADS client. The existing `targetIp` configuration key is retained, so saved configurations do not need migration. Schema/regression checks cover the change; DNS resolution still depends on the installation's network. This is available in the fork and has not been merged into the original adapter.

Proposed upstream in [PR #445](https://github.com/iobroker-community-adapters/ioBroker.beckhoff/pull/445). [Successful fork CI](https://github.com/zonfacter/ioBroker.beckhoff/actions/runs/38462698504). Upstream integration/closure remains with the maintainers.

mcm1957, 2026-10-14:
Thanks for providng PR. Please process review remarks

mcm1957, 2026-10-19:
Implemented with 2.0.0

closed 2026-10-19 by mcm1957